openai/openai-realtime-embedded #2: Performance and Compatibility Issues with Realtime API on ESP32-S3

espressif2022, 2024-12-20:
Hello, I am trying the Realtime API on ESP32-S3. Our implementations on both Python and the ESP32-S3 device are functioning correctly. However, on the device side, the upload and audio download processes are too slow, resulting in suboptimal performance. I still have a few questions:

1. As introduced in the API documentation, only `24kHz` sampled data is supported. However, our device-side local `SR model` supports `16kHz`, and 24kHz incurs a significant overhead for embedded devices. Is it possible to configure this?

2. The previous `Create Speech API` supported formats like MP3, Opus, AAC, FLAC, WAV, and PCM. Does the` conversation.item.create` API support format configuration?

3. I also tested the `openai-realtime-embedded-sdk`. I noticed that it processes `I2S raw data` through `opus_encode` before sending it to the server. However, I find this demo's processing a bit confusing and seemingly inconsistent with the development documentation.

Please forgive my lack of understanding, and I look forward to your reply，thanks.

juberti-oai, 2024-12-27:
The websocket API only accepts 24KHz PCM (and G.711). We recommend using the WebRTC API (as used by this SDK) if you want to transmit compressed formats.

akdeb, 2024-12-29:
If you'd like a full-duplex audio websocket implementation, we have one here for ESP32-S3 https://github.com/StarmoonAI/starmoon

espressif2022, 2024-12-30:
> The websocket API only accepts 24KHz PCM (and G.711). We recommend using the WebRTC API (as used by this SDK) if you want to transmit compressed formats.

ok，thanks. 
I've tested both methods (WebSocket with G.711 and WebRTC API). While using WebRTC, I occasionally experience noisy audio during broadcasts. I'm unsure if this is caused by network issues.

espressif2022, 2024-12-30:
> If you'd like a full-duplex audio websocket implementation, we have one here for ESP32-S3 https://github.com/StarmoonAI/starmoon

Thanks for sharing! I'll take a look and learn from it.

closed 2025-01-06 by espressif2022

akdeb, 2025-04-24:
@espressif2022 We launched a new repo supporting OpenAI Realtime API on the ESP32 with Arduino!! Let me know if it helps with your use case https://github.com/akdeb/ElatoAI