openai/openai-realtime-embedded #6: Raspberry Pi Zero compatibility status

wishtodaya, 2024-12-24:
Hello,

I'm interested in running the OpenAI Realtime Embedded SDK on a Raspberry Pi Zero. I see that the SDK supports Linux platforms, and since Raspberry Pi Zero runs Linux (ARM), I wanted to ask:

1. Has anyone tested this SDK on Raspberry Pi Zero?
2. Are there any known limitations or considerations for ARM-based Linux systems?
3. Would additional setup steps be required beyond those listed in the README?

Thanks for any information about compatibility with RPi Zero!

leeebo, 2024-12-25:
The Linux build is also based on the ESP-IDF build system, please refer https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/host-apps.html?highlight=linux

juberti-oai, 2024-12-27:
This SDK currently assumes ESP32, the audio code would need to be rewritten for Linux.