openai/openai-realtime-embedded #7: Does azure_api_key support this sdk?

xinliqingpi, 2024-12-25:
Does azure_api_key support this sdk?

juberti-oai, 2024-12-27:
Not yet. Should be easy to add to http.cpp.

HaoHoo, 2024-12-28:
add_compile_definitions(OPENAI_API_KEY="$ENV{OPENAI_API_KEY}")
add_compile_definitions(OPENAI_REALTIMEAPI="https://api.openai.com/v1/realtime?model=gpt-4o-mini-realtime-preview-2024-12-17")

Looks like that real-time model in Azure OpenAI uses WebSocket to connect to the service:
wss://my-eastus2-openai-resource.openai.azure.com/openai/realtime?api-version=2024-10-01-preview&deployment=gpt-4o-realtime-preview-deployment-name
(ref: https://learn.microsoft.com/en-us/azure/ai-services/openai/how-to/realtime-audio)

I guess that the model version is different and API is too.

# Backend Selection
BACKEND=azure|openai

# Azure OpenAI Configuration
AZURE_OPENAI_ENDPOINT=your-endpoint
AZURE_OPENAI_DEPLOYMENT=your-deployment

# OpenAI Configuration
OPENAI_API_KEY=your-api-key
OPENAI_MODEL=your-model

juberti-oai, 2024-12-30:
I think Azure just needs some slightly different HTTP headers, e.g., `x-api-key` instead of `authorization`, and a different base URL (openai.azure.com/blah instead of api.openai.com/blah)

HaoHoo, 2025-01-10:
> I think Azure just needs some slightly different HTTP headers, e.g., `x-api-key` instead of `authorization`, and a different base URL (openai.azure.com/blah instead of api.openai.com/blah)

Looks like Azure OpenAI only supports Secure WebSocket now.