openai/openai-realtime-embedded #9: Messages over datachannel are ignored

fedorenchik, 2024-12-26:
I'm trying to send `session.update` message over the WebRTC data channel with `peer_connection_datachannel_send()` but never get message back and model parameters seem to be not changing. Looks like it doesn't work?

juberti-oai, 2024-12-27:
Is receiving messages over the datachannel working?

GOROman, 2025-01-03:
The peer_connection_create_datachannel added at https://github.com/sepfy/libpeer/pull/176 in libpeer will allow messages to be returned from WebRTC DataChannel.

seaduboi-openai, 2025-01-05:
Hi @fedorenchik this is fixed in master!

If enabled we print all datachannel messages [here](https://github.com/openai/openai-realtime-embedded-sdk/blob/main/src/webrtc.cpp#L33)

By default on connect we do a [response.create](https://github.com/openai/openai-realtime-embedded-sdk/blob/main/src/webrtc.cpp#L43-L44) to demonstrate how to send messages.

Try it out! If you have any other ideas on how to make the SDK better we would love to hear!

closed 2025-01-05 by seaduboi-openai