openai/openai-realtime-embedded #13: unable to connect tp AP

POWERFULMOVES, 2024-12-30:
im getting an error  on the  freenove esp32 s3 
i followed instructions but it wont connect 
W (1402) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
I (3822) realtimeapi-sdk: retry to connect to the AP
I (3822) realtimeapi-sdk: connect to the AP fail
I (6232) realtimeapi-sdk: retry to connect to the AP
I (6232) realtimeapi-sdk: connect to the AP fail
I (8632) realtimeapi-sdk: retry to connect to the AP
I (8632) realtimeapi-sdk: connect to the AP fail
I (11042) realtimeapi-sdk: retry to connect to the AP
I (11042) realtimeapi-sdk: connect to the AP fail
I (13452) realtimeapi-sdk: retry to connect to the AP
I (13452) realtimeapi-sdk: connect to the AP fail
I (15862) realtimeapi-sdk: connect to the AP fail

MaskSigner, 2024-12-31:
if your wifi name and password is correct,please check if your wifi is 2.5G ,it can't connect 5G

makerawei, 2025-03-04:
I got the same error and I'am sure that my wifi-ssid and password are correct.  
And the WiFi works on 2.4GHz.  
Have you fixed the error? @POWERFULMOVES

POWERFULMOVES, 2025-03-04:
> I got the same error and I'am sure that my wifi-ssid and password are correct.  
> And the WiFi works on 2.4GHz.  
> Have you fixed the error? @POWERFULMOVES 

No i think its something that has to be changed on the router and i didnt bother to dig into it the instruction that came with the device was able to connect to my router just fine but the code in this sdk doesnt

makerawei, 2025-03-05:
> No i think its something that has to be changed on the router and i didnt bother to dig into it the instruction that came with the device was able to connect to my router just fine but the code in [...]

Thank you for your response. I tried using a different development board, and the WiFi connection issue disappeared. It might be that my original development board was faulty. Additionally, do you know if this SDK supports the INMP441 digital microphone?

PoohWoah, 2025-03-06:
> > > I got the same error and I'am sure that my wifi-ssid and password are correct.我遇到了同样的错误，但我确信我的 wifi-ssid 和密码是正确的。
> > > And the WiFi works on 2.4GHz.并且 WiFi 工作在 2.4GHz。
> > > Have you fixed the error? [@POWERFULMOVES](https://github.com/POWERFULMOVES)您修复错误了吗？ [@POWERFULMOVES](https://github.com/POWERFULMOVES)
> > 
> > 
> > No i think its something that has to be changed on the router and i didnt bother to dig into it the instruction that came with the device was able to connect to my router just fine but the code in this sdk doesnt不，我认为这是路由器上需要更改的东西，我没费心去深入研究它，设备附带的说明可以很好地连接到我的路由器，但此 sdk 中的代码却不能
> 
> Thank you for your response. I tried using a different development board, and the WiFi connection issue disappeared. It might be that my original development board was faulty. Additionally, do you know if this SDK supports the INMP441 digital microphone?感谢您的回复。我尝试使用不同的开发板，WiFi 连接问题消失了。可能是我原来的开发板有问题。另外，您知道这个 SDK 是否支持 INMP441 数字麦克风吗？

Hello, may I ask how the response speed of this sdk is? Under the condition of good network, I want to try this sdk at present

makerawei, 2025-03-06:
> Hello, may I ask how the response speed of this sdk is? Under the condition of good network, I want to try this sdk at present

Although the WiFi connection was successful and I received the model's greeting, there seems to be an issue with my INMP441 microphone. The model doesn't appear to be receiving my voice data, so I'm unable to test the latency. Does anyone know how to resolve the microphone issue?

PoohWoah, 2025-03-06:
> Although the WiFi connection was successful and I received the model's greeting, there seems to be an issue with my INMP441 microphone. The model doesn't appear to be receiving my voice data, so I'm [...]

Did you develop with esp32s3？