openai/openai-realtime-embedded #17: ICE connection negotiation fails when attempting to establish WebRTC connection.

yuchen-0520, 2025-01-06:
### Logs

```
I (2589) realtimeapi-sdk: got ip:10.1.20.41
I (2589) esp_netif_handlers: sta ip: 10.1.20.41, mask: 255.255.255.0, gw: 10.1.20.254
INFO    ./deps/libpeer/src/agent.c      32      create IPv4 UDP socket: 54
INFO    ./deps/libpeer/src/peer_connection.c    175     Datachannel allocates heap size: 102400
INFO    ./deps/libpeer/src/peer_connection.c    182     Audio allocates heap size: 8096
I (2699) realtimeapi-sdk: PeerConnectionState: new
I (5899) realtimeapi-sdk: HTTP_EVENT_DISCONNECTED
I (5899) realtimeapi-sdk: PeerConnectionState: checking
I (35909) realtimeapi-sdk: PeerConnectionState: failed
```

LiuZeyu1119, 2025-01-08:
Same question, do you use the transit api？

LiuZeyu1119, 2025-01-08:
static void oai_on_icecandidate_task(char *description, void *user_data) {
  char local_buffer[MAX_HTTP_OUTPUT_BUFFER + 1] = {0};
  oai_http_request(description, local_buffer);
  ESP_LOGI("oai_on_icecandidate_task", "description: %s", description);
  peer_connection_set_remote_description(peer_connection, local_buffer);
} 

## I used the transit api and noticed that the ip address is 0.0.0.0, which is probably wrong: description: 

v=0
o= -1495799811084970 1495799811084970 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic: iot
a=group:BUNDLE audio datachannel
m=audio 9 UDP/TLS/RTP/SAVP 111
a=rtpmap:111 opus/48000/2
a=ssrc:6 cname:webrtc-opus
a=sendrecv
a=mid:audio
c=IN IP4 0.0.0.0
a=rtcp-mux
a=fingerprint:sha-256 89:AC:33:11:99:3C:7A:61:A2:28:E9:C4:AB:3E:23:2E:11:E7:90:DC:D6:BC:09:64:83:62:1B:12:C2:62:FE:38
a=setup:passive
a=ice-ufrag:8L4q
a=ice-pwd:8L4qayzgTV7AdZTvYfbXRoj5
a=candidate:0 1 UDP 2127783423 192.168.14.253 54117 typ host
m=application 50712 UDP/DTLS/SCTP webrtc-datachannel
a=mid:datachannel
a=sctp-port:5000
c=IN IP4 0.0.0.0
a=max-message-size:262144
a=fingerprint:sha-256 89:AC:33:11:99:3C:7A:61:A2:28:E9:C4:AB:3E:23:2E:11:E7:90:DC:D6:BC:09:64:83:62:1B:12:C2:62:FE:38
a=setup:passive
a=ice-ufrag:8L4q
a=ice-pwd:8L4qayzgTV7AdZTvYfbXRoj5
a=candidate:0 1 UDP 2127783423 192.168.14.253 54117 typ host

jkl375, 2025-01-09:
> ### Logs
> ```
> I (2589) realtimeapi-sdk: got ip:10.1.20.41
> I (2589) esp_netif_handlers: sta ip: 10.1.20.41, mask: 255.255.255.0, gw: 10.1.20.254
> INFO    ./deps/libpeer/src/agent.c      32      create [...]

I have the same error, do you solve it?
```
E (1702529411) esp-tls: [sock=4] select() timeout
E (1702529411) esp-tls: Failed to open new connection
E (1702529411) transport_base: Failed to open a new connection
E (1702529411) HTTP_CLIENT: Connection failed, sock < 0
E (1702529411) realtimeapi-sdk: Error perform http request ESP_ERR_HTTP_CONNECT
I (1702529411) realtimeapi-sdk: HTTP_EVENT_DISCONNECTED
I (1702529411) realtimeapi-sdk: PeerConnectionState: checking
I (1702529426) realtimeapi-sdk: PeerConnectionState: failed
```

nischalj10, 2025-01-13:
@jkl375 

to solve the error -
```
E (1702529411) esp-tls: [sock=4] select() timeout
E (1702529411) esp-tls: Failed to open new connection
E (1702529411) transport_base: Failed to open a new connection
E (1702529411) HTTP_CLIENT: Connection failed, sock < 0
E (1702529411) realtimeapi-sdk: Error perform http request ESP_ERR_HTTP_CONNECT
```

remove 
```
CONFIG_ESP_TLS_INSECURE=y
CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y
```
and add
```
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=y
```

to your `sdkconfig.defaults` file 

and add 
```
config.timeout_ms = 20000;
config.crt_bundle_attach = esp_crt_bundle_attach;
```

to your esp_http_client in http.cpp

doesn't fully solve the issue and I intermittently still face it but much more reliable.

dustMason, 2025-01-15:
I was experiencing this issue (PeerConnectionState never reached `completed`), but when I tested the same code/device on a different wi-fi network, it worked reliably. I suggest investigating your network configuration.

seaduboi-openai, 2025-03-25:
Development for ESP32 has moved to the implementation provided by Espressif. [esp-webrtc-solution](https://github.com/espressif/esp-webrtc-solution/tree/main/solutions/openai_demo

If anyone has issues in the future that is the best place to ask. Thanks for trying this out!

closed 2025-03-25 by seaduboi-openai