openai/openai-realtime-embedded #27: An Enhanced Version of OpenAI WebRTC Demo for ESP32 Series Board

TempoTian, 2025-01-26:
We are excited to announce an enhanced version of the OpenAI WebRTC demo, specifically designed for the ESP32 series platform! 

You can explore it here: [OpenAI WebRTC Demo for ESP32](https://github.com/espressif/esp-webrtc-solution/tree/main/solutions/openai_demo). This version brings advanced features to improve performance and user experience. 

## ✨ Key Enhancements:  
- **Enhanced Peer Connection:**  
  - Improved `esp_peer` implementation with support for TURN, RTP NACK, DataChannel SACK, and more.  
- **Optimized Codec Performance:**  
  - Leverages `esp_audio_codec` for efficient OPUS encoding and decoding, ensuring superior audio quality.  
- **Improved Interaction Quality:**  
  - Integrates Acoustic Echo Cancellation (AEC) for clearer and more reliable voice interactions.  
- **Integrated, Ready-to-Use Solution**  
  - Eliminating the need to build media system components from scratch.  

## 🙏 We’d Love Your Feedback!  
We encourage you to:  
- Explore this enhanced version.  
- Share your feedback and suggestions for future improvements.  
- Leave your feedback in this issue or submit it directly to the official repository.  

Your insights are incredibly valuable, and we look forward to hearing from you!  

📹 **Demo Video**:  
We’ve recorded a short demo showcasing chat and voice-command-triggered function calls. Check it out here:  
[Watch the Demo](https://www.youtube.com/shorts/EraiYGlp28E)  

Thank you for your time and support! 💬

nischalj10, 2025-01-26:
This is awesome! Thank you @TempoTian

seaduboi-openai, 2025-01-26:
This is amazing @TempoTian!

Would you be ok if I deprecated this repo and directed users to the espressif example?

Or would it be better to copy that new code into this repo (and preserve authors)

Thanks for making this so much easier for developers

TempoTian, 2025-02-05:
@seaduboi-openai  Apologies for the delayed response. In my opinion, it would be beneficial to keep this repository as it also supports non-Espressif platforms. Additionally, consider adding a note in the README to guide users to the new example for the Espressif platform.

seaduboi-openai, 2025-03-25:
I have updated the README to direct users to the new repo, thank you for for making it @TempoTian!

In the future I will work on supporting more platforms. At this time ESP32 is the only thing I have been testing/using it on.

closed 2025-03-25 by seaduboi-openai

akdeb, 2025-04-24:
In case people are looking for an Arduino solution we have one here https://github.com/akdeb/ElatoAI
Just bring your own OPENAI_API_KEY and get talking :-)